rwb27/openflexure_nano_motor_controller #2: Break out more pins

rwb27, 2017-06-28:
I’ve been thinking a bit more about this, and we might struggle for pins after all – Currently we have 14 channels of Darlington pair drivers (of which 12 are used).  Extra things we might like are:

1.	Limit switches (up to 3)
2.	LED control (using Darlington pair or, better, i2c-controlled digital potentiometers with constant-current drivers)
3.	Heater control (using one or two analogue pins plus a PWM pin for output, driving a power MOSFET)

The Nano has 14 digital pins of which 2 are the serial interface and 12 control the motors.  That leaves us 8 analogue pins to play with, which would be just enough to do the 3 things above, either using a digipot + constant current circuit for the LEDs or just the darlington pairs.

rwb27, 2017-10-03:
I think the two unused Darlington pairs are broken out now - though I don't know if that version has been uploaded (?)

rwb27, 2017-11-28:
I'm going to make a number of more specific issues for the next revision of the board.

rwb27, 2017-11-29:
[image]
While there are lots of nice suggestions for the next version, I think these are the most useful changes in the short term, and shouldn't be too much work...

sctv, 2017-12-11:
As few more suggestions.... for the future.

For LEDs you could use neopixels or similar.

https://create.arduino.cc/projecthub/whimsy-makerspace/arduino-compatible-nano-neopixel-controller-6f0c4b

On the nano analogue pins 1-6 can be used as digital if required ;)

"A0-A5 can be used as: analog input/digital input/digital output.
A6 and A7 are analog inputs only." (Via https://forum.arduino.cc/index.php?topic=502105.0

rwb27, 2017-12-13:
Thanks - that is useful.  I've shied away from the neopixels for the moment, largely because my experience with other multi-colour LEDs is that the colours come from different locations on the die - while that's not an issue in lighting/display applications, it can be a real pain in microscopy.  

That said, we are looking at more sophisticated condenser designs that might include diffusers, at which point neopixels might be a very good option :)

sctv, 2017-12-14:
Its good to know about that disadvantage with multi led dies + microscopy.

sangavalerian, 2018-01-17:
thanks for the comments, i am finalizing version 0.2 of sangaboard (openflexure_nano_motor_controller). This time i have used eagle.