rwb27/openflexure_nano_motor_controller #14: Include ATMega 328

rwb27, 2017-11-28:
If we move to surface mount components, we could include the microcontroller (and other bits that make a Nano, i.e. the crystal, power supply, CH340 serial chip, and ISP connector).  This would make it smaller and cheaper.

codeTom, 2018-07-11:
Would this really be cheaper than the arduino? Cheapest Nano (unsoldered headers, on alibaba) is $0.34/pcs+shipping, atmega+ch340+quartz from cheapest alibaba suppliers are around $0.3 and unless it's assembled in china, shipping from different suppliers will likely make it much more expensive than buying the nano directly.

rwb27, 2018-07-19:
It's not so much about expense as about convenience - if we integrate the surface-mount components, then you don't need to solder on the Arduino - that means less assembly for us, and a smaller overall package (as you don't need to waste space for two boards stacked on top of each other or include 2.54mm headers).

This only makes sense if we're having the board populated - if we are assembling it ourselves, it almost certainly is better just to solder on a Nano.

rwb27, 2019-02-27:
This was replaced with a more capable microcontroller.

closed 2019-02-27 by rwb27