rwb27/openflexure_nano_motor_controller #20: kitspace requires a README.md

rwb27, 2018-03-08:
I changed README.md to README.rst for compatibility with Sphinx, but now I've broken compatibility with kitspace.org.  Perhaps we should split the repository...

sctv, 2018-03-28:
Can you have both?

kasbah, 2018-03-28:
Oops, didn't see this. I have added limited .rst support to Kitspace (but you could probably have both too).

rwb27, 2018-07-05:
Thanks - I think that fixed the issue, I've pulled your PR now :)

rwb27, 2018-07-05:
I accepted the PR, but it seemed not to fix it.  I should just split the Python readme and the repository readme.md, and possibly add a separate one for kitspace anyway.

kasbah, 2018-07-05:
I believe you can now have a markdown readme for PyPI: https://dustingram.com/articles/2018/03/16/markdown-descriptions-on-pypi

kasbah, 2018-07-05:
Ah, I see. Should have read your comment properly. Anyway, like I said. We have limited .rst support (not yet previewed on kitspace.org/submit) so it's ok for now: http://add-openflexture.preview.kitspace.org/boards/github.com/kitspace-forks/openflexure_nano_motor_controller/