rwb27/openflexure_nano_motor_controller #30: Make light sensor support a run-time decision

rwb27, 2018-07-20:
Currently support for the light sensor has to be compiled in.  It would be great to be able to set this at runtime, and I can't see why that's a bad idea...

codeTom, 2018-07-25:
The obvious drawback is it will use more program memory (with both adafruit libraries we're at ~65%, help strings take a lot), it also adds some global variables, ~300 extra bytes (we would still have ~960 bytes left), both of these should be fine now, but if more light sensors or other features are added, we might run out of memory.
Whoever is building the microscopes will need to flash the firmware anyway, so selecting the configuration they are building should not be a problem (we can separate the config to a separate file if that makes it easier).

rwb27, 2018-07-25:
yeah, that's probably why it's a bad idea.  It might save a few seconds for lazy people (AKA me) but it's not a big problem.  It's not like flashing the firmware is hard in any case.  I think I'll close this one for now.